bxparks/AceButton #20: Example Request

jgandt, 2019-02-23:
Hi There,

I love your library and it's interface.

I haven't written C++ in years, and I would love an analog button read example.

Roughly following the capacitive example, I think it has something to do with overriding `ace_button::ButtonConfig::readButton`. I have attempted it but I'm doing something wrong.

I think I'm doing all of the subclassing properly, but as I said I haven't written C++ in a loooong time.

Definitely feel free to close as won't-fix. Just thought I'd throw it out as something that might be useful to users.

Thanks!

bxparks, 2019-02-24:
Hi, thanks. But I don't understand your question. How do you want to use AceButton for an analog button? It wasn't designed for an analog button. It assumes that you have a digital button.

bxparks, 2019-03-11:
If no further comments, I will close this during my next housekeeping sweep.

jgandt, 2019-03-11:
Yeah sorry so I have a specific use case where I want to translate an analog signal from a button press into a discrete set of values. Basically I am taking an analog signal and converting it to an integer in the range of 1-5 depending on the signal.

I just want to be able to debounce the button press, regardless of the value. Basically, if the value is < x, you can consider the button unpressed.

It seems like it should be very similar to debouncing a binary button press (Low/Hi), but I was unable to get it working.

In any case, I have gone with a different package, so please feel free to close this.

Thanks!

bxparks, 2019-03-18:
It sounds like you needed a custom `ButtonConfig` that converted the analog signal into a binary signal. Which seems doable. But I'm glad you found a solution that works for you.
Closing.

closed 2019-03-18 by bxparks