bxparks/AceButton #25: Does not build on Arduino Nano Every

seisfeld, 2019-06-27:
Hi,

building AceButton on the brand new [Arduino Nano Every](https://store.arduino.cc/nano-every) (megaAVR architecture) fails with this error message:

```
/libraries/AceButton/src/ace_button/AceButton.cpp:33:4: error: #error HIGH must be defined to be 1
   #error HIGH must be defined to be 1
    ^~~~~
exit status 1
```

It's obviously this piece of code:

```
// Check that the Arduino constants HIGH and LOW are defined to be 1 and 0,
// respectively. Otherwise, this library won't work.
#if HIGH != 1
  #error HIGH must be defined to be 1
#endif
#if LOW != 0
  #error LOW must be defined to be 0
#endif
```

I tracked down where `HIGH` and `LOW` are defined (`hardware/arduino/avr/cores/arduino/Arduino.h`):

```
#define HIGH 0x1
#define LOW  0x0
```

This looks fine to me. I removed the check, which clears the error message and the builds but the library does not work. So as advertised in the comment above the check haha. Any idea how to resolve this?

If you need more information, let me know.

bxparks, 2019-06-27:
Hi,
My guess is that `HIGH` is *not* defined by `hardware/arduino/avr/cores/arduino/Arduino.h`, because it's defined to be `0x1` on my Arduino Nano (non-Every), and it works fine.

I don't have a Nano Every board. Can you create a short program to just print out your `HIGH` value? Basically, something like:
```c++
void setup() {
  Serial.begin(115200);
  while (!Serial);
  Serial.print("HIGH: ");
  Serial.println(HIGH);
}
```

seisfeld, 2019-06-27:
Sure, I tried that already. It prints

HIGH: 1

and

LOW: 0

respectively. Which made me google this and I found that location on stack overflow.

bxparks, 2019-06-27:
I just realized that I can compile without actually owning one of these boards, so I installed the 'Arduino megaAVR Boards 1.8.1" core, and discovered it in `hardware/megaavr/1.8.1/cores/arduino/api/Common.h`:

```C++
typedef enum {
  LOW     = 0,
  HIGH    = 1,
  CHANGE  = 2,
  FALLING = 3,
  RISING  = 4,
} PinStatus;
```

This is unfortunate, because that means that `HIGH` is not a C prepropcessor macro but is an enum value. This also explains the compiler warnings that you should be seeing:
```C++
/home/brian/dev/arduino-1.8.9/portable/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/Common.h:104:6: note:   initializing argument 2 of 'void digitalWrite(pin_size_t, PinStatus)'
 void digitalWrite(pin_size_t pinNumber, PinStatus status);
      ^~~~~~~~~~~~
/home/brian/dev/AceButton/examples/HelloButton/HelloButton.ino:51:36: warning: invalid conversion from 'int' to 'PinStatus' [-fpermissive]
       digitalWrite(LED_PIN, LED_OFF);
```

I'm not sure what the solution is. Let me think about this...

bxparks, 2019-06-27:
Related info at https://github.com/arduino/ArduinoCore-API/issues/25

seisfeld, 2019-06-27:
Allright thank you. For the time being I'll swap my regular Nano back in my circuit. 😉 I have a couple more libraries not working in my project, so I guess this board is just too new. I mean it was literally shipped by Arduino this week...

If you need my to test anything or need more sample code actually run, let me know. I am happy to help.

bxparks, 2019-06-27:
Changing the signature of `digitalWrite()` to use a `PinStatus` enum seems like a really bad idea. But AceButton should be able to support this, since the semantics of `HIGH` did not change. I'll look into it.

bxparks, 2019-06-28:
Hi,

I'm driving blind here, since I don't have this board to test on. Can you try changing in `AceButton.h`, line 80:

```C++
static const uint8_t kButtonStateUnknown = 2;
```
to
```C++
static const uint8_t kButtonStateUnknown = 5
```

and see if that works? You'll have to comment out the `#error` of course. (And when you wrote "does not work" above, what did you mean more precisely?)

seisfeld, 2019-06-29:
Hi Brian,

there was some evidence discovered in the the last hours that there - in addition to the PinStatus change - is actually a bug in the Arduino megaAVR Boards 1.8.1. I opened an issue [1] there because the simplest things (like digitalRead and digitalWrite) did not work at all on the analog pins (my current project uses A0-A2 for button inputs). On the digital pins both worked fine. I have been told that this is known and will be fixed sometime next week. They gave me a beta of 1.8.2 and I went ahead and tested AceButton with that one.

Good news: It appears everything works as expected in AceButton - when i comment out the `#error` of course. I built a small setup on a breadboard, just one button and the SingleButton example works fine.

So in summary, the only thing you need to check is how to deal with the changes you mentioned [here](https://github.com/bxparks/AceButton/issues/25#issuecomment-506472218). The reason why it still did not work after commenting out the `#error` was the bug mentioned in [1] and will be fixed soon. A change of `kButtonStateUnknown` was not required.

[1] https://github.com/arduino/Arduino/issues/9024

seisfeld, 2019-07-01:
megaAVR-1.8.2 has been released today.

bxparks, 2019-07-05:
Hi Stephan,
Can you test the latest on the `develop` branch?
I believe I made this fully compatible with the Nano Every and other megaAVR boards.

seisfeld, 2019-07-05:
Hi Brian,

I did run a few quick tests (just the Every and one button) on my breadboard - works like charm. Awesome! I did test the SingleButton, HelloButtom and AutoBenchmark examples with no change (except the button pin, I changed that to A0). I will swap the Every into my project this weekend again and see if it works there as well (which I do expect).

If you want me to run specific tests for you, please let me know. Thanks for the quick turnaround!

cheers
Stephan

bxparks, 2019-07-06:
Thanks for testing. I just released version 1.3.4 with this fix. Closing. Let me know if you run into any problems.

closed 2019-07-06 by bxparks