bxparks/AceButton #29: Multiple Buttons

GavinBrownAus, 2019-09-14:
Hi there, just a quick question. Do you have a good example using AceButton for multiple independent buttons that are floating that go low when pressed. All are connected to individual pins on a teensy. 

I've tried a number of the examples but they dont seem to work as described. 

Thanks

bxparks, 2019-09-14:
Buttons can't be floating, they need to have a pull-up or a pull-down resistor (internal or external). If you didn't mean to write that, then take a look at examples/TunerButtons and examples/ArrayButtons.

(I won't be able to respond to followups for about 2 weeks...)

bxparks, 2019-10-01:
Back from vacation. Do you have any further questions? Otherwise, I'm going to close this.

bxparks, 2019-10-12:
Closing due to lack of activity.

closed 2019-10-12 by bxparks