bxparks/AceButton #38: Analog Pins without GPIO

DarthBrento, 2020-02-22:
Hi,

I had some trouble with one of the multiple buttons of my project not working. It took me a while until I found out why. I am using almost all PINs on my Arduino Nano and ended up using A6, which does not have the GPIO functionality. It would be nice if this issue and possibly the fix were mentioned somewhere in the README.md.
I fixed it with adding a pull-up resistor (10k) and replacing following code in ButtonConfig.h:

```    
    virtual int readButton(uint8_t pin) {
      if (pin == A6 || pin == A7)
        return (analogRead(pin) >= 512 ? HIGH : LOW);
      return digitalRead(pin);
    }
```

It works reliably for me, even with the standard setup `pinMode(A6, INPUT_PULLUP);`. So with an external pull-up, the above code is the only change needed and those pins behave like any other. Are there any issues with this solution? Except of course you have to match the pins to your hardware if you do not use a Nano.

Regards,
Ben

bxparks, 2020-02-23:
Hi,
I never intended this library to work with analog pins, and have never tested it on them. I *think* your solution is ok.. but I'm not really sure. 
* One problem might be that `analogRead()` takes longer than `digitalRead()`, [this page](https://www.arduino.cc/reference/en/language/functions/analog-io/analogread/) says ~100us, so not too bad.
* The other problem is that some analog pins have 10-bit resolution, and some have 12-bit resolution, some may have different resolutions. So your 512 mid-point value needs to change accordingly.
* Another problem is that on some micro-controllers, A6 and A7 are perfectly valid digital pins (e.g. SparkFun Pro Micro), so this isn't a general solution.

I recommend creating a subclass of `ButtonConfig` and override `readButton()` in your subclass. This avoids having to maintain a slightly different version of AceButton for yourself.

If you are using lots of buttons and running out of GPIO pins, this binary encoding technique  (https://github.com/bxparks/AceButton/blob/develop/docs/binary_encoding/README.md) that I recently added may be of use to you.

Have fun!

DarthBrento, 2020-02-24:
I took over your library from some other code, where it was already used on the analog pins, so initially, I did not know what it was intended for and only came to the repository when things did not work as expected. I opened the "issue" only to discuss and share my solution, not to criticize anything.  

Thanks for your recommendation, I will do it that way. Your binary encoding is not an option, because I want to detect simultaneous key presses.

bxparks, 2020-02-24:
No worries, I did not interpret anything you wrote as critical. Just out of curiosity, what do you mean you "took over from some other code". Are you guys using my library in a commercial product?

DarthBrento, 2020-02-26:
I saw your library there: https://github.com/seisfeld/TonUINO

I do something similar but with my own code. So far it is only a private project.

closed 2020-02-26 by DarthBrento