bxparks/AceButton #40: GPIO pin 0 cannot be used?

henk65, 2020-03-24:
Trying to use AceButton on pin 0 gives me this error:

  *exit status 1*
  *call of overloaded 'AceButton(int)' is ambiguous*

All other pins are okay, except pin 0. What can be done about this?

bxparks, 2020-03-24:
Try `AceButton((uint8_t) 0)`.

This is caused by 2 overloaded constructors:

* `AceButton(uint8_t pin, ...)` and
* `AceButton(ButtonConfig* buttonConfig, ...)`

C++ maps a `0` to either an `int` or a `void*`. I'm not really sure how to get around that. Open to suggestions.

bxparks, 2020-03-24:
This stackoverlow suggests a way to disambiguate (https://stackoverflow.com/questions/4610503/how-to-resolve-ambiguity-of-call-to-overloaded-function-with-literal-0-and-point) but it's kinda ugly. Not sure I want to introduce that kind of clutter into the code.

henk65, 2020-03-24:
Okay, it's working now. That's not the way I usually do it, but it did the trick for me and I'm fine with it.

Shall I close this issue now, or do you want to leave it open for suggestions from others? You can close it if you want to, I'm okay with that...

bxparks, 2020-03-24:
Added notes in the README.md.

closed 2020-03-24 by bxparks

bxparks, 2020-03-24:
It occurred to me that an alternative, perhaps cleaner, solution is to use a const:
```C++
const uint8_t PIN = 0;
AceButton button(PIN);
```