bxparks/AceButton #45: Interrupts

pruwait, 2020-06-20:
How can I use interrupts with this libruary, like in EasyButton?

bxparks, 2020-06-21:
This library does not officially support interrupts. It's probably possible to add some flags to the ISR handler, then call the AceButton::check() method outside the ISR. But it's not something that I want to support officially, at least at this moment.

bxparks, 2020-06-27:
To be more clear, I'm not 100% sure what you mean by interrupts in this case. This library does not interfere with something else using interrupts. The library does not depend on interrupts. What should the button do when the interrupt is triggered? We still need to debounce the button, because the bouncing will cause multiple interrupts to be fired from a single push of a button. So we still need to poll the button for 20 ms. In any case, there is no direct link between this library and interrupts, so I'm going to close this issue.

closed 2020-06-27 by bxparks