bxparks/AceButton #48: Compiling on Nano Every gives errors

PhilHaw, 2020-07-13:
On trying to compile, for example, HelloButton for one of the new Nano Every modules, the Arduino IDE spits out a load of errors (see below). These seem to be related to the testing code in your library. I am using Version 1.5.0 of your AceButton library and Arduino IDE 1.8.13. Every other board type I've tried seems to compile without any issues.

Arduino: 1.8.13 (Windows 10), Board: "Arduino Nano Every, ATMEGA328"

In file included from F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:2:0:

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.h:62:18: error: reference to 'Print' is ambiguous

     void printTo(Print& printer) const;

                  ^~~~~

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.h:30:7: note: candidates are: class Print

 class Print;

       ^~~~~

In file included from C:\Users\Philip\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/deprecated/Print.h:23:0,

                 from F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:1:

c:\users\philip\appdata\local\arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino\api\print.h:34:7: note:                 class arduino::Print

 class Print

       ^~~~~

In file included from F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:2:0:

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.h:62:18: error: 'Print' has not been declared

     void printTo(Print& printer) const;

                  ^~~~~

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.h:98:18: error: reference to 'Print' is ambiguous

     void printTo(Print& printer) const;

                  ^~~~~

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.h:30:7: note: candidates are: class Print

 class Print;

       ^~~~~

In file included from C:\Users\Philip\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/deprecated/Print.h:23:0,

                 from F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:1:

c:\users\philip\appdata\local\arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino\api\print.h:34:7: note:                 class arduino::Print

 class Print

       ^~~~~

In file included from F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:2:0:

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.h:98:18: error: 'Print' has not been declared

     void printTo(Print& printer) const;

                  ^~~~~

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:7:27: error: variable or field 'printTo' declared void

 void EventRecord::printTo(Print& printer) const {

                           ^~~~~

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:7:27: error: reference to 'Print' is ambiguous

In file included from F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:2:0:

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.h:30:7: note: candidates are: class Print

 class Print;

       ^~~~~

In file included from C:\Users\Philip\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/deprecated/Print.h:23:0,

                 from F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:1:

c:\users\philip\appdata\local\arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino\api\print.h:34:7: note:                 class arduino::Print

 class Print

       ^~~~~

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:7:34: error: 'printer' was not declared in this scope

 void EventRecord::printTo(Print& printer) const {

                                  ^~~~~~~

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:7:34: note: suggested alternative: 'printf'

 void EventRecord::printTo(Print& printer) const {

                                  ^~~~~~~

                                  printf

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:17:28: error: variable or field 'printTo' declared void

 void EventTracker::printTo(Print& printer) const {

                            ^~~~~

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:17:28: error: reference to 'Print' is ambiguous

In file included from F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:2:0:

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.h:30:7: note: candidates are: class Print

 class Print;

       ^~~~~

In file included from C:\Users\Philip\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/deprecated/Print.h:23:0,

                 from F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:1:

c:\users\philip\appdata\local\arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino\api\print.h:34:7: note:                 class arduino::Print

 class Print

       ^~~~~

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:17:35: error: 'printer' was not declared in this scope

 void EventTracker::printTo(Print& printer) const {

                                   ^~~~~~~

F:\Arduino\libraries\AceButton\src\ace_button\testing\EventTracker.cpp:17:35: note: suggested alternative: 'printf'

 void EventTracker::printTo(Print& printer) const {

                                   ^~~~~~~

                                   printf

exit status 1

Error compiling for board Arduino Nano Every.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

bxparks, 2020-07-15:
I think I fixed it but I cannot be 100% sure, because I don't own a Nano Every. Hope this works for you.

closed 2020-07-15 by bxparks

PhilHaw, 2020-07-15:
Thanks Brian, that now compiles on the Nano Every and the HelloButton sketch works.

I’ll let you know if I come across any other issues. It seems the Every will have broken quite a few libraries ☹

Many thanks for the great AceButton library – it really works well, has great features and is very easy to use.

Regards,

Philip

From: Brian Park
Sent: 15 July 2020 05:46
To: bxparks/AceButton
Cc: PhilHaw; Author
Subject: Re: [bxparks/AceButton] Compiling on Nano Every gives errors (#48)

I think I fixed it but I cannot be 100% sure, because I don't own a Nano Every. Hope this works for you.

commit b25e0dec (2020-07-15): src/ace_button/testing/EventTracker.h: Remove unnecessary forward declaration for 'Print'; apparently breaks 'Nano Every', because the megaAVR toolchain breaks backward compatibility with all other previous Arduinos (fixes #48)