bxparks/AceButton #49: Build AceButton with AUniter

victorhg, 2020-08-21:
Hi!

I'm experimenting with AceButton, and trying to build it using the AUniter cli for AUnit... I'm sure this is a Path problems, but don't know how to solve it:

`~/development/AceButton % auniter verify esp32 $(find . -name '*.ino')                                                 [0]
======== Processing environment 'esp32'
-------- Processing file './tests/EncodedButtonConfigTest/EncodedButtonConfigTest.ino'
$ /Applications/Arduino.app/Contents/MacOS/Arduino --verify --board esp32:esp32:esp32:PartitionScheme=default,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none --pref 'compiler.cpp.extra_flags=-DAUNITER -DAUNITER_ESP32 -DAUNITER_SSID="wonderland2G_EXT" -DAUNITER_PASSWORD="a1b2c3d4e5"' ./tests/EncodedButtonConfigTest/EncodedButtonConfigTest.ino
2020-08-20 22:36:33.970 Arduino[57775:1921892] Loading Application 'Arduino'
2020-08-20 22:36:33.971 Arduino[57775:1921892] JVMRuntime=jre8u252-b09.jre
2020-08-20 22:36:33.971 Arduino[57775:1921892] CFBundleName=Arduino
2020-08-20 22:36:33.971 Arduino[57775:1921892] WorkingDirectory=(null)
2020-08-20 22:36:33.971 Arduino[57775:1921892] JVMMainClassName=processing.app.Base
2020-08-20 22:36:33.971 Arduino[57775:1921892] JVMOptions=(
    "-Dapple.awt.application.name=Arduino",
    "-Dcom.apple.macos.use-file-dialog-packages=true",
    "-Dcom.apple.smallTabs=true",
    "-DAPP_DIR=$APP_ROOT/Contents/Java",
    "-Djava.ext.dirs=$JVM_RUNTIME/Contents/Home/lib/ext/:$JVM_RUNTIME/Contents/Home/jre/lib/ext/",
    "-Djava.net.preferIPv4Stack=true",
    "-Xdock:name=Arduino",
    "-Dcom.apple.mrj.application.apple.menu.about.name=Arduino",
    "-Dfile.encoding=UTF-8",
    "-Xms128M",
    "-Xmx512M",
    "-splash:$APP_ROOT/Contents/Java/lib/splash.png"
)
2020-08-20 22:36:33.972 Arduino[57775:1921892] JVMArguments=(
)
2020-08-20 22:36:33.972 Arduino[57775:1921892] JVMClasspath=(null)
2020-08-20 22:36:33.972 Arduino[57775:1921892] JVMDefaultOptions={
}
2020-08-20 22:36:33.972 Arduino[57775:1921892] -> Bundle path: /Applications/Arduino.app
2020-08-20 22:36:33.972 Arduino[57775:1921892] -> Working Directory: '/Users/victorhg/development/AceButton'
2020-08-20 22:36:33.972 Arduino[57775:1921892] -> JVM Runtime path: /Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre
2020-08-20 22:36:33.972 Arduino[57775:1921892] Searching for a Java 8 virtual machine
2020-08-20 22:36:33.972 Arduino[57775:1921892] Search for java VM in '/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home'
2020-08-20 22:36:33.972 Arduino[57775:1921892]   KO - error: 'launch path not accessible'
2020-08-20 22:36:34.041 Arduino[57775:1921892] No matching JDK found.
2020-08-20 22:36:34.042 Arduino[57775:1921892] -> Java Runtime Dylib Path: '/Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre/Contents/Home/lib/jli/libjli.dylib'
2020-08-20 22:36:34.056 Arduino[57775:1921892] Command line passed to application argc=27:
2020-08-20 22:36:34.057 Arduino[57775:1921892] Arg 0: '/Applications/Arduino.app/Contents/MacOS/Arduino'
2020-08-20 22:36:34.057 Arduino[57775:1921892] Arg 1: '-Djava.class.path=/Applications/Arduino.app/Contents/Java/log4j-core-2.12.0.jar:/Applications/Arduino.app/Contents/Java/rsyntaxtextarea-3.0.3-SNAPSHOT.jar:/Applications/Arduino.app/Contents/Java/batik-ext-1.8.jar:/Applications/Arduino.app/Contents/Java/slf4j-api-1.7.22.jar:/Applications/Arduino.app/Contents/Java/arduino-core.jar:/Applications/Arduino.app/Contents/Java/batik-xml-1.8.jar:/Applications/Arduino.app/Contents/Java/slf4j-simple-1.7.22.jar:/Applications/Arduino.app/Contents/Java/batik-dom-1.8.jar:/Applications/Arduino.app/Contents/Java/commons-compress-1.8.jar:/Applications/Arduino.app/Contents/Java/commons-httpclient-3.1.jar:/Applications/Arduino.app/Contents/Java/bcprov-jdk15on-152.jar:/Applications/Arduino.app/Contents/Java/commons-logging-1.0.4.jar:/Applications/Arduino.app/Contents/Java/log4j-api-2.12.0.jar:/Applications/Arduino.app/Contents/Java/batik-script-1.8.jar:/Applications/Arduino.app/Contents/Java/jackson-annotations-2.9.5.jar:/Applications/Arduino.app/Contents/Java/batik-parser-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-squiggle-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-awt-util-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-gvt-1.8.jar:/Applications/Arduino.app/Contents/Java/bcpg-jdk15on-152.jar:/Applications/Arduino.app/Contents/Java/xml-apis-1.3.04.jar:/Applications/Arduino.app/Contents/Java/batik-anim-1.8.jar:/Applications/Arduino.app/Contents/Java/jtouchbar-1.0.0.jar:/Applications/Arduino.app/Contents/Java/batik-bridge-1.8.jar:/Applications/Arduino.app/Contents/Java/jssc-2.8.0-arduino4.jar:/Applications/Arduino.app/Contents/Java/batik-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-transcoder-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-svg-dom-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-rasterizer-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-codec-1.8.jar:/Applications/Arduino.app/Contents/Java/commons-io-2.6.jar:/Applications/Arduino.app/Contents/Java/commons-codec-1.7.jar:/Applications/Arduino.app/Contents/Java/xmlgraphics-commons-2.0.jar:/Applications/Arduino.app/Contents/Java/apple.jar:/Applications/Arduino.app/Contents/Java/commons-exec-1.1.jar:/Applications/Arduino.app/Contents/Java/batik-css-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-util-1.8.jar:/Applications/Arduino.app/Contents/Java/jna-platform-4.2.2.jar:/Applications/Arduino.app/Contents/Java/commons-net-3.3.jar:/Applications/Arduino.app/Contents/Java/batik-svgpp-1.8.jar:/Applications/Arduino.app/Contents/Java/xml-apis-ext-1.3.04.jar:/Applications/Arduino.app/Contents/Java/jmdns-3.5.5.jar:/Applications/Arduino.app/Contents/Java/java-semver-0.8.0.jar:/Applications/Arduino.app/Contents/Java/pde.jar:/Applications/Arduino.app/Contents/Java/jna-4.2.2.jar:/Applications/Arduino.app/Contents/Java/commons-lang3-3.8.1.jar:/Applications/Arduino.app/Contents/Java/jackson-core-2.9.5.jar:/Applications/Arduino.app/Contents/Java/jsch-0.1.50.jar:/Applications/Arduino.app/Contents/Java/jackson-databind-2.9.5.jar'
2020-08-20 22:36:34.057 Arduino[57775:1921892] Arg 2: '-Djava.library.path=/Applications/Arduino.app/Contents/MacOS'
2020-08-20 22:36:34.057 Arduino[57775:1921892] Arg 3: '-DLibraryDirectory=/Users/victorhg/Library'
2020-08-20 22:36:34.057 Arduino[57775:1921892] Arg 4: '-DDocumentsDirectory=/Users/victorhg/Documents'
2020-08-20 22:36:34.057 Arduino[57775:1921892] Arg 5: '-DApplicationSupportDirectory=/Users/victorhg/Library/Application Support'
2020-08-20 22:36:34.057 Arduino[57775:1921892] Arg 6: '-DCachesDirectory=/Users/victorhg/Library/Caches'
2020-08-20 22:36:34.058 Arduino[57775:1921892] Arg 7: '-DSandboxEnabled=true'
2020-08-20 22:36:34.058 Arduino[57775:1921892] Arg 8: '-Dapple.awt.application.name=Arduino'
2020-08-20 22:36:34.058 Arduino[57775:1921892] Arg 9: '-Dcom.apple.macos.use-file-dialog-packages=true'
2020-08-20 22:36:34.058 Arduino[57775:1921892] Arg 10: '-Dcom.apple.smallTabs=true'
2020-08-20 22:36:34.058 Arduino[57775:1921892] Arg 11: '-DAPP_DIR=/Applications/Arduino.app/Contents/Java'
2020-08-20 22:36:34.058 Arduino[57775:1921892] Arg 12: '-Djava.ext.dirs=/Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre/Contents/Home/lib/ext/:/Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre/Contents/Home/jre/lib/ext/'
2020-08-20 22:36:34.058 Arduino[57775:1921892] Arg 13: '-Djava.net.preferIPv4Stack=true'
2020-08-20 22:36:34.058 Arduino[57775:1921892] Arg 14: '-Xdock:name=Arduino'
2020-08-20 22:36:34.058 Arduino[57775:1921892] Arg 15: '-Dcom.apple.mrj.application.apple.menu.about.name=Arduino'
2020-08-20 22:36:34.059 Arduino[57775:1921892] Arg 16: '-Dfile.encoding=UTF-8'
2020-08-20 22:36:34.059 Arduino[57775:1921892] Arg 17: '-Xms128M'
2020-08-20 22:36:34.059 Arduino[57775:1921892] Arg 18: '-Xmx512M'
2020-08-20 22:36:34.059 Arduino[57775:1921892] Arg 19: '-splash:/Applications/Arduino.app/Contents/Java/lib/splash.png'
2020-08-20 22:36:34.059 Arduino[57775:1921892] Arg 20: 'processing.app.Base'
2020-08-20 22:36:34.059 Arduino[57775:1921892] Arg 21: '--verify'
2020-08-20 22:36:34.059 Arduino[57775:1921892] Arg 22: '--board'
2020-08-20 22:36:34.059 Arduino[57775:1921892] Arg 23: 'esp32:esp32:esp32:PartitionScheme=default,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none'
2020-08-20 22:36:34.059 Arduino[57775:1921892] Arg 24: '--pref'
[... cut ...]
2020-08-20 22:36:34.081 Arduino[57775:1921902] Arg 26: './tests/EncodedButtonConfigTest/EncodedButtonConfigTest.ino'
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
EncodedButtonConfigTest:4:23: fatal error: AceButton.h: No such file or directory
compilation terminated.
exit status 1
FAILED verify: esp32  ./tests/EncodedButtonConfigTest/EncodedButtonConfigTest.ino`

Should be able to work out of the box, right? But it seems not to find the headers contained on src folders.

Thanks

bxparks, 2020-08-21:
Looks like you are running AUniter from MacOS, but MacOS is not supported. See https://github.com/bxparks/AUniter:
"Some exploration on MacOS has been done, but it is currently not supported." 
It can probably be made to work with GNU utilities installed through `brew`, but I don't want to spend the time on it since I don't use my Mac for development anymore.

It's possible that AUniter can be migrated to be built on top of [Arduino CLI](https://github.com/arduino/arduino-cli) instead, but I haven't had the time to look into that yet.

If you haven't done so, I suggest making sure that the program compiles under the Arduino IDE (instead of through AUniter), just to make sure that you got the various dependencies and libraries installed properly.

bxparks, 2020-08-24:
I pulled out my mac laptop this weekend, and I managed to get `auniter.sh` working on my Mac. But note that I run 10.14 (Mojave), I don't have a Mac that runs 10.15 (Catelina). I updated the Installation instructions for the Mac in the [auniter/README.md](https://github.com/bxparks/AUniter/tree/develop/tools). You need to install the GNU coreutils and GNU gsed packages using `brew`.

I also managed to get `auniter.sh` working with [arduino-cli](https://github.com/arduino/arduino-cli), on both Linux and MacOS. The instructions are in the README.md referenced above.

If you want to give this another shot, check out the latest version from the 'develop' branch on GitHub.

With regards to your compilation error:
```
EncodedButtonConfigTest:4:23: fatal error: AceButton.h: No such file or directory
compilation terminated.
exit status 1
```
this usually means that the `AceButton/` folder is not included in the sketchbook `./libraries/` directory. Are you sure that the Arduino IDE was able to compile `EncodedButtonConfigTest.ino`? If the IDE is able to compile it manually, then `auniter.sh` should be able to compile it as well.

bxparks, 2020-09-07:
I'm going to close this, since it is not directly related to the AceButton library. If you have any additional comments/questions about AUniter, you can file a ticket on that project. Hope you got enough useful information. Good luck!

closed 2020-09-07 by bxparks