bxparks/AceButton #62: Capacitive button returns release while button is pressed

GHBLoos, 2020-12-19:
I am using the capacitive button as a switch to control a smart light. The problem is that the released occurs while not expected. I don't understand why and whether this is AceButton related or CapacitiveSensor related. 

**Hardware**
- Wemos D1 mini
- D2 (GPIO 4) connected to wire and metal strip
- D1 (GPIO 5) connected with resistor (680k) to D2

**Problem**
When touching and holding the metal strip I expect that the buttonevent is _pressed_. To see what event occurs, I addds a `Serial.Println` with `millis()` and the `eventType`. The monitor shows:
```
368761 Pressed
368792 Clicked
368792 Released
368821 Pressed
368852 DoubleClicked
368852 Released
368881 Pressed
368912 Clicked
368913 Released
368941 Pressed
368972 DoubleClicked
368972 Released
369001 Pressed
369032 Clicked
369032 Released
369062 Pressed
369094 DoubleClicked
369094 Released
```
So while I expected to only see a _pressed_, I also see a _released_, and because of that also _clicked_ and _doubleClicked_.

I am not sure why this occurs:
* bad hardware setup
* wrong capacitiveSensor config
* something in the code

**Code**
The code is used to function as a smart (led) light, which is controlled by diyHue https://diyhue.github.io/. This light has a touch button to switch the light on or off.

The full code is below (based upon https://github.com/diyhue/Lights/tree/master/Arduino/Generic_SK6812_Strip, by Marius Motea).

```
// diyHue switch 
// AceButton with capacitiveSensor
// ESP 8266 (Wemos D1 min)
// WS2812 RGBW led on D4

// diyHue must be installed (on a RPi for example)
// after first start, ESP functions as a WIFI access point (AP). Connect to the access point (for example with tablet or phone) and use your browser to go to 192.168.4.1 
// to set wifi credentials

#include <FS.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ESP8266HTTPUpdateServer.h>
#include <ESP8266WebServer.h>
#include <NeoPixelBus.h>
#include <WiFiManager.h>
#include <ArduinoJson.h>

#include <CapacitiveSensor.h>
#include <AceButton.h>
using namespace ace_button;

/**
 * A subclass of ButtonConfig that allows a CapacitiveSensor to emulate a
 * mechanical switch connected to a pull-up resistor on the input pin. A "touch"
 * sends a LOW signal, just like a mechnical switch.
 */
[... 904 lines cut ...]
              strip->SetPixelColor(pixel, blendingEntert(lights[0].currentColors, lights[1].currentColors, pixel + 1 - (lightLedsCount - transitionLeds / 2 )));
            }
          }
        } else if (light == lightsCount - 1) {
          for (uint8_t pixel = 0; pixel < lightLedsCount - transitionLeds / 2 ; pixel++) {
            strip->SetPixelColor(pixel + transitionLeds / 2 + lightLedsCount * light, convInt(lights[light].currentColors));
          }
        } else {
          for (uint8_t pixel = 0; pixel < lightLedsCount; pixel++) {
            if (pixel < lightLedsCount - transitionLeds) {
              strip->SetPixelColor(pixel + transitionLeds / 2 + lightLedsCount * light, convInt(lights[light].currentColors));
            } else {
              strip->SetPixelColor(pixel + transitionLeds / 2 + lightLedsCount * light, blendingEntert(lights[light].currentColors, lights[light + 1].currentColors, pixel - (lightLedsCount - transitionLeds ) + 1));
            }
          }
        }
      } else {
        strip->ClearTo(RgbwColor(lights[0].colors[0], lights[0].colors[1], lights[0].colors[2], 0), 0, lightLedsCount - 1);
      }
    }
    strip->Show();
  }
}

```

bxparks, 2020-12-19:
Hi,
Since I don't have your hardware, I can only guess. But I'm pretty sure that it's not AceButton, it's been used and tested by too many people to have a fundamental bug like this. It's probably something to do with the CapacitiveSensor library and how it behaves when you keep touching it, and what it does when it reaches its `TIMEOUT_MILLIS`. Something is causing it to behave as though you are touching it and releasing it rapidly, over 10-20 milliseconds, which causes AceButton to think that you are clicking and double clicking.

You will have difficulty debugging this with all the extra cruft in your code. What you should do is to create a new program, and strip this down to the bare essence, removing everything that does not involve AceButton or CapacitiveSensor. Then insert some print statements inside the `CapacitiveSensor.capacitiveSensor()` function, and figure out why it thinks you are touching and removing your finger on the switch.

bxparks, 2021-01-02:
I'm going to close this, since it does not seem like a bug in AceButon.

closed 2021-01-02 by bxparks