bxparks/AceButton #64: [OT] featuring AceButton

ubidefeo, 2021-01-09:
hi Brian

I have no other means to get in touch with you (no social media or contacts are in your profile), so here is the only place I can post a message to you.
I wanted to let you know I have used AceButton in our content update video on Arduino CLI 0.14.0
https://www.youtube.com/watch?v=vmTYh9jDoQ8

you can remove this issue now :)
regards,

ubi [ Arduino Tooling Team ]

bxparks, 2021-01-11:
Hello Ubi,
I thought my email was public on GitHub but it looks like it was not. I have fixed it. I guess people were able to email me directly before by looking at the bottom of my `README.md` files, or getting the email from the `library.properties` files. I deliberately do not have any social media presence. I think that comes from having a healthy concern about the erosion of privacy in our current digital world, and knowing what is currently done with millions of computers with access to exabytes of data.

Thank you for mentioning AceButton in your tutorial. I found it useful, especially now that I have started to use it more after you fixed the `--build-property` flag. I noticed that you were using the `arduino:samd` core version 1.8.9 in your tutorial. That version uses the old Arduino API, which has been fully tested with AceButton. However, version 1.8.10 migrates to the new Arduino API, which breaks backwards compatibility with most of my libraries. It does not break the core AceButton library code directly, but it does break AUnit which is how I do unit testing of AceButton.

I wrote a long comment about this breakage here:
https://github.com/arduino/ArduinoCore-samd/issues/587#issuecomment-751406035
You can tell that I have become frustrated about the backwards incompatible breakages of the new Arduino API.

I was just about to start blacklisting platforms which I no longer wish to support, starting with the arduino:megaavr and the arduino:samd cores. Then I received this note from you, and I guess I feel obligated to avoid breaking your tutorial for your users. So I'm going to describe my attempt to workaround the Arduino API breakages in the https://github.com/arduino/ArduinoCore-samd/issues/587 ticket. I will mention you in it, so that you can have a reference to my thoughts.

closed 2021-01-11 by bxparks

ubidefeo, 2021-01-11:
> arduino/ArduinoCore-samd#587 (comment) <https://github.com/arduino/ArduinoCore-samd/issues/587#issuecomment-751406035>
> I was just about to start blacklisting platforms which I no longer wish to support, starting with the arduino:megaavr and the arduino:samd cores. Then I received this note from you, and I guess I feel obligated to avoid breaking your tutorial for your users. So I'm going to describe my attempt to workaround the Arduino API breakages in the arduino/ArduinoCore-samd#587 <https://github.com/arduino/ArduinoCore-samd/issues/587> ticket. I will mention you in it, so that you can have a reference to my thoughts.

hey Brian

I didn’t think about looking into the library.properties file :D
thanks for taking the time to get in touch.

The breaking APIs are an issue here and there, in fact sometimes we have to patch things because it also breaks out stuff.
I guess there’s a bit of a transition period, things inside Arduino are taking a way more structured shape as our team grows.

I wouldn’t worry too much for now about accommodating just because of the new CLI video.
In the end it’s not really a tutorial, but I used your repo to display this new feature allowing users to install a library from a Git repo directly.
I would nonetheless like to have people like you support our efforts and get onboard with the framework.
I really invite you to voice your opinion, by now you know we listen.
Also regarding `--build-property` I'm still not happy with the awkwardness of it and we're investigating nicer ways to add custom `-D` flags.
I am one of the advocates for it, so you can count on it becoming a thing :D

I am going to carefully read your long comment and understand what your issues are.
I hope you'll want to consider keeping compatibility with the framework.
Also the work we're doing with the new tool `arduino-lint` is gonna be of big help to library and platform developers when it comes to raising red (or orange) flags

Keep up the great work,
hope to be crossing paths again :)

ciao.ubi

ubi de feo
hybrid designer

Future Tailors

Nieuwpoortkade 2a
1055RX Amsterdam
The Netherlands
m	+31 6 46027890

http://ubidefeo.com
@ubidefeo