bxparks/AceButton #120: use of  __FlashStringHelper is ambiguous

geoffreyheller, 2023-05-25:
Hello,

Love your AceButton library!

One minor issue with the 1.10.0 release:   In AceButton.h you define class __FlashStringHelper in the global namespace, but in AceButton.cpp you define it in the ace_button namespace.

Depending on how restrictive your compiler is, this may result in a compiler error.  In my case I'm seeing this:
`.pio/libdeps/debug/AceButton/src/ace_button/AceButton.h:127:12: error: reference to '__FlashStringHelper' is ambiguous`

The issue can be fixed by moving __FlashStringHelper into the ace_button namespace in AceButton.h

geoffreyheller, 2023-05-25:
That's a possibility.   I can comment out line 26 in Acebutton.h and I can build ok
`class __FlashStringHelper;`
It is possible this means __FlashStringHelper is defined elsewhere.

That said, it is ambiguous to be referencing __FlashStringHelper on lines 79 and 83 in AceButton.cpp while in the ace_button namespace while defining the __FlashStringHelper in AceButton.h in the global namespace.

If that inconsistency is intentional then perhaps add a comment explaining why it is important.

bxparks, 2023-05-25:
For some reason, GitHub sent an email with your previous reply, but there is no record of it here:

> That's a possibility. I can comment out line 26 in Acebutton.h and I can build ok
> class __FlashStringHelper;
> It is possible this means __FlashStringHelper is defined elsewhere.
>
> That said, it is [...]

I think you are confused about the code in `AceButton.cpp`. All of that code is within the `ace_button::` namespace, so there is no additional redeclaration of `__FlashStringHelper`.

The problem is that ArduinoCore-API redefines `__FlashStringHelper` (and all other previous Arduino classes into the new `arduino::` namespace, then adds a `using arduino::__FlashStringHelper` statement to pollute the global namespace anyway, in an attempt to provides backwards compatibility, which then causes conflicts with any other previous application-level declarations which previously used the previously valid declaration of `__FlashStringHelper` in the global namespace.

Removing the statement in `AceButton.h` is the actual solution here, because I forgot that I include `#include <Arduino.h>` at the top of that file, so I don't need my own declaration of `__FlashStringHelper`. I will make that change shortly.

closed 2023-05-25 by bxparks

bxparks, 2023-05-25:
Released v1.10.1 with this fix. Enjoy!

geoffreyheller, 2023-05-25:
Thank you for taking care of this so quickly!

commit 8bf6a635 (2023-05-25): AceButton.h: remove unnecessary declaration of __FlashStringHelper, already provided by Arduino.h (fixes #120)