bxparks/AceButton #122: invalid use of non-static member function 'void Interface::handleEvent(ace_button::AceButton*, uint8_t, uint8_t)'

Sercurio, 2024-02-21:
Hello,

I try to instantiate a ButtonConfig in a member class function but I get the following error.

It is possible to do this pattern ?

```Test.h
class Test {
    private:
        AceButton button1 = AceButton(1);
        void configure();
};
```

```Test.cpp
#include <Test.h>

void Test::configure(){
     pinMode(1, INPUT_PULLUP);
     ButtonConfig *buttonConfig = ButtonConfig::getSystemButtonConfig();
     buttonConfig->setEventHandler(handleEvent);
     buttonConfig->setFeature(ButtonConfig::kFeatureClick);
     buttonConfig->setFeature(ButtonConfig::kFeatureLongPress);
     buttonConfig->setFeature(ButtonConfig::kFeatureDoubleClick);
     buttonConfig->setFeature(ButtonConfig::kFeatureRepeatPress);
}

void Test::handleEvent(AceButton *button, uint8_t eventType, uint8_t buttonState){
     switch (button->getPin())
    {
    case 1:
        Serial.println("Hourrray");
        break;
    }
}
```

Thank you

bxparks, 2024-02-21:
Try declaring your `Test::handleEvent()` as `static`. The `AceButton::setEventHandler()` expects a pointer to a plain function, not a pointer to a member function.

Sercurio, 2024-02-22:
Yes I tried, but the problem is that I use member variable in my eventHandler function like that:

class Test {
    private:
        AceButton button1 = AceButton(1);
        int counter = 0;
        void configure();
        void handleEvent(AceButton *, uint8_t, uint8_t);
};

#include <Test.h>

void Test::configure(){
     pinMode(1, INPUT_PULLUP);
     ButtonConfig *buttonConfig = ButtonConfig::getSystemButtonConfig();
     buttonConfig->setEventHandler(handleEvent);
     buttonConfig->setFeature(ButtonConfig::kFeatureClick);
     buttonConfig->setFeature(ButtonConfig::kFeatureLongPress);
     buttonConfig->setFeature(ButtonConfig::kFeatureDoubleClick);
     buttonConfig->setFeature(ButtonConfig::kFeatureRepeatPress);
}

void Test::handleEvent(AceButton *button, uint8_t eventType, uint8_t buttonState){
     switch (button->getPin())
    {
    case 1:
        counter+=1;
        break;
    }
}